betaflight/blackbox-log-viewer #8: Maxing out CPU

Arakon, 2016-06-07:
I'm sorry that this report will be fairly vague, but I have no more details than this.

- When switching between several logs (16 MB usually), leaving the BB viewer open for a while, switching between hotkey displays, etc, eventually scrolling by grabbing the trace view or throttle graph with the mouse and pulling left or right will randomly freeze first the app itself and any other chrome apps for several seconds, with my CPU fan audibly increasing speed (indicating high load).
- Eventually this results in most programs on the PC freezing for up to 20 seconds (i.e. VLC).
- During this, although it may be a seperate issue, sometimes switching from one sub-log within the same file results in no traces being shown and the throttle graph not being updated. Only restarting the app helps.

- Restarting the app fixes the freezes until they occur again a while later
- Using the original BB viewer, no freezes ever happen

Note that this is an I5 PC with plenty of power, not a laptop or similar that may actually easily be overloaded.

GaryKeeble, 2016-06-07:
@Arakon Thanks for the feedback report, I have managed to replicate the issue on my machine and it is related to the spectrum analyser feature grabbing an audio context every-time the log file is changed (Chrome has a limit of 6 times before crashing...). 

I have patched the viewer to cater for this so the problem should be resolved. If you could try it please and let me know if you still have issues.

Arakon, 2016-06-07:
I just messed around with it some and it appears to be fixed. I didn't do any detailed testing, but if it occurs again during everyday use, I'll reopen this.
Thanks for the quick fix.

closed 2016-06-07 by Arakon