betaflight/blackbox-log-viewer #11: Release Management

borisbstyle, 2016-08-02:
@GaryKeeble 
Keep track of versioning in manifest file. I am not putting it into chrome store.

GaryKeeble, 2016-08-03:
@borisbstyle 
No problem;

I have only added a link to the welcome page that looks at github for the latest release and shows a button with the current latest release in it. User's can then click button to go to the github 'releases' page and download it if they want to. Isn't truly release management but will help users know when something new is available.

Can remove it completely if you prefer...

borisbstyle, 2016-08-03:
@GaryKeeble 
Its no problem at all. I was just notifying you that its in chrome store from now :D

GaryKeeble, 2016-08-03:
Oh; right; excellent. :-)

closed 2016-08-15 by GaryKeeble