betaflight/blackbox-log-viewer #13: Spectrum analyzer display size on changing log entries

etracer65, 2016-08-13:
Running the latest version directly from github (commit 518a636).

If the spectrum analyzer window is maximized, selecting a different log resets the display to the small window but the button stays in the maximized state.  So clicking the button doesn't do anything because the window is already small.  Just looks like the window shouldn't revert in size when selecting a new log.

GaryKeeble, 2016-08-13:
Yes; I can replicate this bug; thanks for finding it and letting me know.

As a short-term workaround; a second press of the button reverts the system to normal behaviour (until you open another log with the analyser maximised that is);

GaryKeeble, 2016-08-14:
This is now fixed in the development branch; It will be included in version 2.5.2

closed 2016-08-15 by GaryKeeble