betaflight/blackbox-log-viewer #14: Feature Request: Show frequency hover on mouse position over analyzer

etracer65, 2016-08-13:
It would be really helpful to see the frequency the mouse is over to help determine cutoffs.  The analyzer is automatically finding the center of the motor noise which is good, but it's a little tough to determine the cutoff.

GaryKeeble, 2016-08-14:
Thanks for the suggestion...
I have added a marker line that shows the frequency of the mouse position as you move over the analyser window whilst holding down the **SHIFT** key. This is currently in the development branch only but will be part of version 2.5.2.

etracer65, 2016-08-14:
I took a look at the current development version and it looks great!  I really like how the marker stays rather than just be a hover value - great idea.  This makes it so much easier to find the cutoff values for the notch filter.

closed 2016-08-15 by GaryKeeble