betaflight/blackbox-log-viewer #17: Feature Request: Adjustable frequency rang in analyzer

etracer65, 2016-08-14:
Sorry, not trying to spam you :)

It would be really useful to be able to adult the min/max frequency range (horizontal axis) to allow more precise analysis of the waveform.  Often the stuff beyond 300hz is not useful but the display going all the way to half the sample frequency (I presume) causes the interesting stuff to scale quite small.

GaryKeeble, 2016-08-14:
Everyone's suggestions are always welcome :)

I must admit; i've wanted to do something like this for a while; I think, for simplicity, i'll try to add the maximum range to the user settings dialog box to make it remember your preference. And your right at the moment the scale is automatically half of the log sample frequency.

etracer65, 2016-08-14:
Sounds good.  They only concern might be that is its a saved setting people might not be able to see any oddities in the spectrum when opening a new log.  You may be interested primarily in the lower frequency stuff, but you might miss some high frequency spike.

Alternately the display could always default to the 1/2 sample range and have some interface to adjust the end point (slider?).

It's a tradeoff between seeing the entire range but always having to adjust, vs. the settings "sticking" but possibly missing something important in the spectrum unless you remember to check (and then you're back to always adjusting the settings anyway).

GaryKeeble, 2016-08-14:
Your absolutely right, slider is the best way to go....

GaryKeeble, 2016-08-14:
I have added two additional slider controls to the fullscreen analyser window. These should allow you to zoom in the frequency range and vertical height of the data.

etracer65, 2016-08-14:
It looks like there are some bugs.  If I move either slider, the graph disappears and simply has "undefined" in the bottom right.  Also the frequency range always jumps to 1/4 the sample rate even when moving the vertical slider.  Resizing the window restores the graph but it seems to return to the default settings (1/2 sample and default scaling) regardless of the sliders.  This is on OS X Chrome if that makes any difference.

etracer65, 2016-08-14:
Hold on - this might be me.  On my main machine I'm still running an old version of the OS and Chrome is only on a 49.x version.  On a fully up to date machine it seems to be working fine.  So probably some compatibility issue with the older Chrome version.  Not sure if this can (or should) be addressed.

etracer65, 2016-08-14:
Sorry - never mind!  It looks like it was a caching issue.  I deleted the old directory and re-cloned the repository but didn't remove it completely from Chrome (delete the app).  Apparently one or more of the javascript files were cached.  Completely removing the app and reinstalling the unpacked extension fixed the problem - even on the old version of Chrome.

GaryKeeble, 2016-08-14:
I'll keep a note of that one; I never realised it cached the files; but it's good to know just in case another user has an issue.

etracer65, 2016-08-15:
Found a small bug.  If you hide the analyzer while it's maximized, the sliders don't go away and stay in the main window.

Also unrelated, another small cosmetic issue.  The tool tip for the open log file button reappears after opening the file and stays.  You can also replicate by clicking cancel in the file dialog.

GaryKeeble, 2016-08-15:
I have fixed the analyser sliders;

The tooltip issue is an odd one; I have noticed this but it uses a standard tooltip facility offered by 'bootstrap' (which all the themes for the whole app are done with). It's not only the file open etc, but many (if not all of the tooltips). The tooltip goes away if you click elsewhere on the page (or hover over another tooltip)... I'll look into this to see if there is a way to stop this behaviour.

borisbstyle, 2016-08-15:
can 2.5.2 be released btw?

GaryKeeble, 2016-08-15:
@borisbstyle 
Yes, I think it would be good to release it and push it to the store as the few extra features on the analyser would certainly be useful for people starting to setup the notch filter in particular;

borisbstyle, 2016-08-15:
@GaryKeeble 
Its released! Thanks for the hard work! 
https://chrome.google.com/webstore/detail/canpiagfkeefejklcojkhojdijglnghc/publish-delayed?hl=en

PS do yourself a favour and put some credits into it!

closed 2016-08-15 by GaryKeeble