betaflight/blackbox-log-viewer #18: Welcome Screen not full seen

TheAngularity, 2016-08-15:
Welcome Screen do not full fill the maximized and reduced windows ... Scrollbar hangs in the air ;-)

maximized window:
[image]

reduced window:
[image]

GaryKeeble, 2016-08-15:
Will look into this; there seems to be some interaction with the screen size when the Viewer is run as an app from the chrome store; I think this is also related to issue #19 too.

TheAngularity, 2016-08-15:
@GaryKeeble 

yep i'm using the viewer as an app

wow, just tested ... you're right .... open the index.html and all ok

TheAngularity, 2016-08-16:
much better now ... thx

closed 2016-08-16 by TheAngularity