betaflight/blackbox-log-viewer #21: Analyser Notch display incorrect when using default settings.

GaryKeeble, 2016-08-15:
The notch filter settings on the analyser display show a filled rectangle covering the range around the notch center value. This should be hidden if the notch_cutoff value is zero (which is the default) otherwise the screen fills the analyser window either side of the notch_center with a semi-transparent rectangle; this is not how it should work when notch_cutoff is zero as this is when the filter is disabled.

@borisbstyle 
This is fixed in v2.5.3 (it's the only change) and I think it may warrant an immediate patch release or it could cause confusion for people just starting to use the analyser with notch settings on default values. (Sorry).

borisbstyle, 2016-08-15:
@GaryKeeble 
No problem. We can bring out 2.5.3 ;)

borisbstyle, 2016-08-15:
https://chrome.google.com/webstore/detail/canpiagfkeefejklcojkhojdijglnghc/publish-delayed?hl=en

closed 2016-08-15 by GaryKeeble