betaflight/blackbox-log-viewer #25: hidden stick display reappears

TheAngularity, 2016-08-16:
hidden stick display reappears when you press shortcut "S", "C" or "G" ... :(

TheAngularity, 2016-08-20:
@GaryKeeble ... have you a solution

GaryKeeble, 2016-08-20:
No solution yet; this will be part of the cleanup of the User Settings mechanism that I am hoping to release as part of the v2.5.4 cleanup.

wolkesson, 2018-04-27:
This is still a bug!!! I've not reached the core why options.drawSticks are not retained when for example "QuickSmooth" is used. But for some reason, the sticks are drawn in the same canvas as the graphs, but the craft and analyzer are drawn in separate ones. The separate canvases are hidden/shown with CSS styles. While the sticks are hidden with a flag in grapher.js. Is there a reason for this inconsistent or would you like me to create a PR with the sticks as a separate canvas, hidden and shown as the others? Would also be nice to split the code into separate files and building components down the line.

McGiverGim, 2018-04-27:
@wolkesson any cleaning in the code is always welcome. If you can do the PR it will be great!

wolkesson, 2018-05-03:
I'm working on a PR for this! Will split out the sticks to separate canvas and a separate js-file. I'll take a couple of days more.

PR #188 (2018-05-27): Split rendering of sticks into separete js-file
The main purpose is to clean up the code. But also fixes #25. This might conflict with #187.
closed 2018-05-27 by mikeller