betaflight/blackbox-log-viewer #26: overlapped log graph

TheAngularity, 2016-08-16:
i'm used the viewer as an chrome app in maximised window ... on first open a new logfile the log graph display is overlapped from the control panel ... but this does not always

[image]

TheAngularity, 2016-08-20:
@GaryKeeble ... here there is also no solution :(

GaryKeeble, 2016-08-20:
I can replicate this problem; and resizing the window after the display is shown clears the issue temporarily. 
I am working on a solution at the moment as part of a cleanup of the UI code that will be part of 2.5.4

GaryKeeble, 2016-09-28:
Just an update on this long-standing bug;

I believe (hope) the issue is now resolved in the next release v2.5.7

TheAngularity, 2016-09-28:
yep ... well seems to be resolved now ;-)

TheAngularity, 2016-10-02:
Ok, I guess the bug is eliminated ... closed now ;)

closed 2016-10-02 by TheAngularity