betaflight/blackbox-log-viewer #28: rcCommand and gyro graphs do not match

IharpI, 2016-09-08:
The numbers in the legend match but the graphs do not although configured the same way.
This way it is impossible to compare the desired rate with the rate of the copter.

[image]
[image]

GaryKeeble, 2016-09-08:
Hi,

The scaling of rcCommand[] graph is always +/-500 (raw values), the value shown in brackets on the legend is converted from this value to the real deg/s so that you can see what it represents. This is how the original log viewer worked and has been left that way for compatibility. (which makes it difficult to compare gyro and rcCommand traces as you've pointed out).

However, I have added a group of pens called rcCommand**s**[] which are shown on the pen selection drop-down at the very end.... these are scaled in deg/s so you can use them to compare rcCommands with gyro and if you have perfect PID's, they should exactly overlay each other. These are the ones I use all the time.

[image]

borisbstyle, 2016-09-09:
Maybe rcCommands should be called commandedRate, angleRate or setpointRate?

IharpI, 2016-09-09:
Thanks for the reply and sorry that I did not find the rcCommand**s**
I just checked the CF viewer and in the latest version the rcCommand matches the behavior of rcCommand**s** in BF log viewer. At least there the rcCommand matches the trace of the gyro pretty good.
Why not just rename rcCommand to something like rcCommandRaw and rcCommand**s** to rcCommand?

GaryKeeble, 2016-09-09:
@borisbstyle Actually, I couldn't think of of a good name for them when I created them (I added 's' for 'scaled', lazy and a naming cop out for sure :-) ). As a old-school controls engineer, I like your term 'setpointRate' as that pretty much says what they are; we could go with that?

@IharpI I didn't realise the original CF viewer had any of the deg/s calculations in it; thanks for letting me know. I think you and Boris are on the same wavelength and a name change is in order.

GaryKeeble, 2016-09-11:
This issue has been resolved in v2.5.6

closed 2016-09-11 by GaryKeeble