betaflight/blackbox-log-viewer #29: Problems reading debug_mode = NOTCH results in Blackbox Explorer

Edgs, 2016-09-11:
I'm not sure where the problem lies, but the values I see in the Betaflight version of Blackbox Explorer for gyro_raw and gyro_notch don't appear correct.  They appear to be the normal gyro values mapped as follows:
gyro_raw[roll] (debug0) = gyro[roll]
gyro_notch[roll] (debug1) = gyro[pitch]
gyro_raw[pitch] (debug2) = gyro[yaw]
gyro_notch[pitch] (debug3) = 0

These were all logged using default filters, and debug_mode = NOTCH
I did notice that the Field values in the log file itself are 'debug' whereas Blackbox Explorer uses the names 'gyro_raw' etc.  Where does it get these names from and how does it know which 'debug' value they should be?  I suspect the problem is in Blackbox Explorer.
If I set debug_mode = NONE then the 'debug' values appear in Blackbox Explorer Field values.

There may be further issues with this as when I changed several filter parameters, and made further logs, all the above traces were zero.

I've attached a short log displaying the problem.
[ONE.TXT](https://github.com/betaflight/betaflight/files/466003/ONE.TXT)

Betaflight firmware version 3.0
Blackbox Explorer version 2.5.6 (would be useful if this was shown in the app!)
Betaflight configurator 1.8.0

GaryKeeble, 2016-09-11:
Are you certain you are using v2.5.6 of the BB viewer; because there are no gyro_raw[roll], gyro_notch[roll] etc parameters shown anymore (was a 2.5.5 thing). The debug fields for debug mode NOTCH are gyro_preNotch[] for each of roll, pitch and yaw.

If you are using defaults for v3.0.0 of BF, then the gyro notch filter is not on by default (only dterm has the notch filter enabled); thus gyro_preNotch[] and gyro[] will be exactly the same values. You need to set the gyro notch to a value before you will see any difference between the debug values and the gyro[] values.

In the BF firmware, the fields are always labelled debug[] (and those are the field labels that actually appear in the logged data); The BB viewer knows which debug mode has been selected and re-labels them to (hopefully) a more meaningful label just for the legend and scaling units etc.

With debug mode NONE, then the values will be flat (or zero).

I have taken a look at your example file; and can confirm that using v2.5.6 to look at the header of the log file; gyro notch is set to 0Hz (i.e. off) thus will not show any difference between the gyro_preNotch[] and gyro[] fields. Only the dterm notch is on which currently has no debug mode to available (i.e. the debug mode NOTCH is only useful for the gyro notch and not dterm notch).

Does this help ?

Edgs, 2016-09-11:
Yes you are correct, I was using the wrong version.  My apologies.  Plus, the version IS shown in BB viewer, I just didn't see it.  Sorry again!
It isn't very clear within Chrome whether it uses the new version or the old version.  I have just followed the same procedure I used before, click on 'Use App' in Chrome store, and now it is using 2.5.6.  So, not a problem with BB viewer.

closed 2016-09-11 by Edgs