betaflight/blackbox-log-viewer #30: Exporting video always Calculating

apeelme, 2016-09-18:
I tried on Debian Jessie Mate, Ubuntu Mate 16.04 (now fresh install) & Chrome 53.0.2785.116 (64-bit) at all times and nothing happens, always _Calculating..._ the file size, and no rendering. Why???

GaryKeeble, 2016-09-18:
Can you confirm that you have loaded a flight video **and** log file before exporting the video ?

The export video feature only works to overlay an existing flight video; you cannot generate a video from log data only.

apeelme, 2016-09-18:
Yes, loaded video and log file before exporting.

GaryKeeble, 2016-09-18:
Looking at the error, it appears that your source video is not opening correctly....
If you look at the following developer tools for a working system, there is another flag called "has-video" which your system does not have.

[image]

It also does not seem to be able to find mode_2.png image file either; I am suspecting some kind of corruption in your app install; could you please remove and re-install the chrome app and see if that clears your issue?

apeelme, 2016-09-18:
I remove app, reinstall nothing changes. Remove google acc, remove app, settings, extension sync, install, nothing changes. Remove (purge) Chrome, install Chromium from ubuntu 16.04 repo, no google acc adding just install BF Explorer..nothing...same error. What's this???

apeelme, 2016-09-18:
GaryKeeble: The _mode_2.png_ problem is solved. In the _user_settings_dialog.js_ file line 233 selecting valid png's with lowercase filename (_mode__), but i download from this repo the source, and go to _images/stick_modes/_ directory, the filenames uppercase (_Mode__). I'm on linux operating system...

accurate information: download this soure, rename png's, launch index.html in Chromium, load log, load video, the devtools say this:
http://imgur.com/a/JEEwf
Click the Export video, and push Begin export button the devtools say:
http://imgur.com/a/yQAnD

(sorry for my english!)

mikeller, 2016-09-18:
Looks like a windows (uppercase === lowercase) <-> anything else (case matters) problem. :-(

GaryKeeble, 2016-09-19:
I will file the lowercase / uppercase filename issue; thank you for finding that.

Do you think it would be possible to provide a download link to both your video and log files that have this problem so that I can investigate further ?

apeelme, 2016-09-19:
OK, attach 2 video, and 1 log file i tried yesterday.
[REC_0008.MOV](https://drive.google.com/open?id=0B3fPjU1mSNINTGFlaTA3RnBsTUE) - RAW Mobius 1080p 30fps file
[first_rolls.mp4](https://drive.google.com/open?id=0B3fPjU1mSNINUkMtdWs5eHJCWWM) - editing RAW Mobius files with KDEnlive, and export x.264 mp4 video
[bb_log5](https://drive.google.com/open?id=0B3fPjU1mSNINal9DSzRSeTVqT3M) - blackbox log file

apeelme, 2016-09-19:
[https://github.com/cleanflight/blackbox-log-viewer/issues/49#issuecomment-247947247](https://github.com/cleanflight/blackbox-log-viewer/issues/49#issuecomment-247947247)

GaryKeeble, 2016-09-19:
Thanks @apeelme; I have just tried your video's and log and for me on OSX everything worked just fine; however, I will take a good look at the commit you've mentioned above; thank you for pointing it out.

mikeller, 2016-09-19:
> The mode_2.png problem is solved. In the user_settings_dialog.js file line 233 selecting valid png's with lowercase filename (mode_), but i download from this repo the source, and go to images/stick_modes/ directory, the filenames uppercase (Mode_). I'm on linux operating system...

@GaryKeeble I am surprised that this works in OSX, but this will definitely not work in linux: If the case of the filename does not exactly match the case of what is being loaded, the file will not be found. (i.e. `mode_2.png` and `Mode_2.png` are two different files in linux.)

GaryKeeble, 2016-09-20:
@mikeller; there are actually two separate issues here, the original issue about the video export just stopping at "Calculating...."; for me on OSX, the video exports fine using @apeelme's video and log files. However, there is a new webm export .js version available in the original BB viewer that has solved this issue for apeelme using linux.

The second issue is the mode_2.png, Mode_2.png that apeelme has also found. You are of course right, the filename is case sensitive on Linux and will need to be fixed...

mikeller, 2016-09-20:
Ah I see. Just wanted to make sure this doesn't get overlooked.

apeelme, 2016-09-20:
I have a small problem, but i don't know open the new issue. After flight save whole flash from my Naze32 rev.5 (i modded to rev6. with 16MB spiflash). When i open the file in the BB no answering...nothing happens. Therefore i use small perl script for split file, and no problem.
You check my LOG what is wrong???

[bblog_tool.zip](https://github.com/betaflight/blackbox-log-viewer/files/483010/bblog_tool.zip)

GaryKeeble, 2016-09-21:
@apeelme I have seen this problem before; if your log file has no extension, the BB viewer doesn't know what type of file you are loading; it could be a video, log or workspace backup so it guesses based upon file size; large logs are incorrectly guessed as video files which it attempts to load; and fails.

The solution is to add a **.txt** extension to your file (blackbox_log_2016-09-17_185150**.txt**). It should just open properly then. I tried this on your file, and it correctly opened after I added the .txt extension.

apeelme, 2016-09-21:
Thats work.... Thanks for your help! ;)

mikeller, 2016-09-21:
@GaryKeeble: Would it make sense to set the default filename suggested by betaflight-configurator to something that is recognised by the log viewer? (Preferrably **not** *.txt, since this will cause windows to try and open it in a text editor if it gets double clicked accidentally, and this can block windows for a long time in the case of large files.)

apeelme, 2016-09-21:
I can see that the file _onboard_logging.js_ line399 accepts TXT extension...this is default extension? But why did not work? I'm on Ubuntu 16.04 Mate.

onboard_logging.js : http://imgur.com/a/w5bRk
save file window: http://imgur.com/a/hGVdl

mikeller, 2016-09-21:
This is just the list of possible extensions. The suggested file name is generated in the lines directly above, and supplied to the function call as `suggestedName'.

https://developer.chrome.com/apps/fileSystem#method-chooseEntry

GaryKeeble, 2016-09-22:
At the moment, the BB viewer recognises 

1. .TXT, .CFL and .LOG as log files, 
2. .AVI, .MOV, .MP4, .MPEG as video files
3. .JSON as workspace backups

If it is not one of those extensions, it takes a guess on the type by file size (if file-size < 10MB then Log file else its a video); of course I could extend the default threshold for the guess as video files are usually huge by comparison.

Personally, I like the .LOG extension 'cause that is what the file is, but I know that on a lot of system's out there, .LOG files are associated with other programs so double clicking would usually open the wrong program.

So perhaps we should add a default of ".BFL" to the configurator for betaflight log?

mikeller, 2016-09-22:
I like `.bfl` (no point screaming the extension ;-)) `.log` is another contentious one, since it is often assumed to be human readable log files, and users doing systems administration will probably have it associated with a log viewer on Windows (I do).

I will change the SDCard logfile writing code in the firmware to create `logXXX.bfl` log files.

GaryKeeble, 2016-09-23:
Thank's @mikeller;

mikeller, 2016-09-23:
Little update, the extension will be `.BFL` (capitals), since the fat implementation we're using only supports capitals.

wolkesson, 2018-05-03:
Are there any remaining sub-issues not addessed yet? Issues above related to extensions are fixed.  and Image filename casing was fixed in 2.5.7. webm-writer version is now 0.1.1. So I guess this can be closed?

McGiverGim, 2018-05-03:
I suppose it has been solved, so I close the issue. If some bug remains it can be opened again.

closed 2018-05-03 by McGiverGim