betaflight/blackbox-log-viewer #31: Feature request: ability to adjust smoothing, zoom, expo on the fly

joshuabardwell, 2016-09-21:
It'd be nice to be able to adjust smoothing, zoom, expo, on the fly. Especially zoom though, Lots of times I wish I could just shift-scroll-wheel or something and zoom a graph larger. This will require a bit of design, as I'm sure sometimes I would want to zoom all graphs and other times maybe just one graph or even one line.  I leave the details of the implementation to the wonderful coders :-)

Thanks for your amazing work on this project BTW.

GaryKeeble, 2016-09-22:
I'll take a look at it Joshua; 

At the moment, the mousewheel just scrolls through time, \<SHIFT\>mousewheel zooms in and out (like the slider at the top) and \<ALT\>mousewheel zooms faster, _but only when the mouse pointer is over the graph_..... I'll see if i can attach an event over the graph legend to allow you to scale vertically using the mousewheel as well.

GaryKeeble, 2016-09-28:
This functionality has been added to the next release (version 2.5.7).

McGiverGim, 2018-04-27:
I think that the shortcuts dialog is a good place for this.

McGiverGim, 2018-05-16:
Closed. Fixed in https://github.com/betaflight/blackbox-log-viewer/pull/155

closed 2018-05-16 by McGiverGim