betaflight/blackbox-log-viewer #38: Gyro Data not Readable

borisbstyle, 2016-12-02:
See 
https://github.com/betaflight/betaflight/issues/1732

@martinbudden 
Blackbox issues can be created here

GaryKeeble, 2016-12-02:
Have modified master to check for gyro_scale as well as gyro.scale (to keep compatibility with old logs). Also, default gyro scaling to 1.0 for BF3.1.0 as scaling is now handled in the flight controller.

TheAngularity, 2016-12-02:
tested your new master ... the logfile i posted looks like correct ... but now old 3.1,0 log files are displayed differently ... i think this is due to the changes in the betaflight code

TheAngularity, 2016-12-02:
the one would be fixed already times ... but vbat issues is still in the room

GaryKeeble, 2016-12-03:
@TheAngularity, Yes, the version check for the gyro scale is based on the version 3.1.0; so old 3.1.0 logs will not display correctly; only 3.1.0 logs from this point forward or logs prior to 3.1.0 will display correctly.

Still investigating vbat scaling.

mikeller, 2016-12-06:
Could the vBat issues be related to the changeover from storing vBatLatest as raw ADC reading to storing it as voltage?

GaryKeeble, 2016-12-07:
@mikeller, thanks, that makes a lot of sense, the logged values hover around 168..148 which is possibly 16.8v to 14.8v.... will take a look....

martinbudden, 2016-12-19:
@GaryKeeble the betaflight PR https://github.com/betaflight/betaflight/pull/1853 now sets the scale correctly, so presumably your change to check the version number can be removed.

GaryKeeble, 2016-12-19:
@martinbudden; OK; will revert the change :-)

TheAngularity, 2016-12-31:
@GaryKeeble 
But still a small question ... how does it look with the vbat issues?

GaryKeeble, 2016-12-31:
Will start investigating the vbat next :-)

GaryKeeble, 2016-12-31:
@TheAngularity, Just pushed a fix for the VBAT scaling; works on the bench.

Happy New Year to you too :-)

TheAngularity, 2016-12-31:
will give it a try ... here another #44

TheAngularity, 2016-12-31:
@GaryKeeble @borisbstyle 
Looks good again now ... thx ... I think It can closed now ;-)

wolkesson, 2018-04-28:
Close this! ;-)

closed 2018-04-28 by wind0r