betaflight/blackbox-log-viewer #39: On DSHOT, motor values do not display properly?

RC-Slater, 2016-12-03:
Please see the attached log.  Why do the motor values show up so differently when using Dshot300?  I'm using one of the latest 3.1 builds and the latest BB viewer.
[LOG00003.BFL.zip](https://github.com/betaflight/blackbox-log-viewer/files/628744/LOG00003.BFL.zip)

GaryKeeble, 2016-12-04:
@RC-Slater Thanks for reporting this one....

@borisbstyle This appears to be related to minthrottle and maxthrottle that is logged to the blackbox; when d-shot is enabled, then there are different values for minthrottle and maxthrottle;
I think that perhaps we need to log the calculated values "motorOutputHigh, motorOutputLow" in mixer.c as minthrottle and maxthrottle as these are the actual active values? This should then allow the viewer to scale the motor[] values correctly.

borisbstyle, 2016-12-06:
@GaryKeeble Dshot has fixed values actually. So when the log viewer knows that dshot is enabled it should know the motor outputs.

GaryKeeble, 2016-12-07:
@borisbstyle 
Yes I saw the hardcoded dshot limits but unfortunately I don't think we can do this in the viewer,

The blackbox.c file uses motorConfig()->minthrottle to encode the motor[] values when writing the log; the value is unsigned so if the calculated dshot 'minthrottle' (motorOutputLow) is less than motorConfig()->minthrottle then the value is clamped to zero and the motor logged value is clipped (which is what I think is happening in the log file above). As a note, I believe this also happens to the rcCommand[THROTTLE] logged value.

 I can put together a PR with the changes for discussion but I don't have a dshot capable rig to test the PR;

mikeller, 2016-12-07:
@GaryKeeble you don't need a full rig to test. If your FC is DShot capable, you can enable `blackbox_on_motor_test`, and then 'simulate' logging a DShot flight by having DShot enabled and using the motors tab in the configurator.

TheAngularity, 2016-12-31:
@GaryKeeble 
any question between dshot throttle and log-seek-bar ... it is normal that the throttle is only displayed from about 40% upwards? ... tested it, but only on the bench

GaryKeeble, 2016-12-31:
The seekbar actually shows the average motor output, not rcCommand[THROTTLE], however, it is still scaled incorrectly when on DSHOT; fix on its way :-)

TheAngularity, 2016-12-31:
@GaryKeeble 
yeep ... dshot scaling now fixed ... thx 👍

wolkesson, 2018-05-03:
This was fixed in commit 4ae98ec68bc2461202bb0668de0b9dcab8151570. This issue can be closed! (Resuming my quest on closing old issues...)

closed 2018-05-03 by McGiverGim