betaflight/blackbox-log-viewer #40: Spec Analyser 'zoom' button stopped working

rodgerramjet, 2016-12-15:
v 2.5.7

A few weeks ago I could zoom in but it was always temperamental - when you changed zoom it 'lost' data till you click on the spec analyser button on the legend again.... it still seems to still have this problem except now you can't work around it by re-opening the spec analyser. Tried on 2 machines.... tried restarting app

[blackbox_log_2016-12-16_084334.TXT](https://github.com/betaflight/blackbox-log-viewer/files/655934/blackbox_log_2016-12-16_084334.TXT)

GaryKeeble, 2016-12-16:
Hello @rodgerramjet, I can open your log file and zoom the analyser view without issue; I am using Macos, Chrome Version 54.0.2840.98 (64-bit); 

The only time I have ever had an issue with the analyser view I had to remove and then re-install the extension, but if your having issues with both machines, then this is probably not going to help.

Can you let me know what you are using please; osx, windows, linux, Chrome version etc...

rodgerramjet, 2016-12-16:
Hi @GaryKeeble 

Chrome 54.0.2840.99 m
Windows 10

I realised chrome was out of date (not sure why it stopped auto updating) - manual update to 55.x and problem gone

Should I close this?

Pity there can't be a warning saying chrome is out of date or something in bbox explorer :)

GaryKeeble, 2016-12-16:
Yeah; warning in the BB explorer would be nice; not sure if I can do it though.

Glad it's working for you now...

I'll close the issue, no worries :-)

closed 2016-12-16 by GaryKeeble