betaflight/blackbox-log-viewer #42: Feature Request - Log Header Information clean up

TheAngularity, 2016-12-30:
can you clean up the log header information to reflect bf 3.1.0 

remove betaflight pid controller 

[image]

and insert the unknown header fields, features osd and feature esc_sensor ... also remove the features oneshot125 and superexpo_rates

[image]

GaryKeeble, 2016-12-30:
Yes,
I'll start to sort these things out; a clean-up has been on my list for some time now :-)

TheAngularity, 2016-12-31:
tested your latest viewer cleanup ...  looks good again 👍

TheAngularity, 2016-12-31:
new request ... you could still move the debug settings from right to left at the bottom ... shortening the log header information output ;-)

TheAngularity, 2016-12-31:
@GaryKeeble 

> new request ... you could still move the debug settings from right to left at the bottom ... shortening the log header information output ;-)

imho this is the last request ... then everything seems to be good ;-))

GaryKeeble, 2016-12-31:
Swapped the Debug Settings over to the left; and moved the "unknown header fields" over to the right as this is usually hidden unless there is a conflict.

TheAngularity, 2016-12-31:
ok ... yeep ... looks very good now ... thx 👍 

wish you a happy New Year ;-)

closed 2016-12-31 by TheAngularity