betaflight/blackbox-log-viewer #48: Adjust 'amperageLatest' data type to be 'SIGNED'

mikeller, 2017-01-22:
Followup from https://github.com/betaflight/betaflight/pull/2192 .

McGiverGim, 2017-05-17:
@mikeller I'm trying to change this in the blackbox viewer. Can you give me some more info? Seeing the PR is not enough for me.
Now, in the log viewer, for BF 3.0 or later, the value informed in the log is readed and divided into 10.
For earlier BF versions there's an ADC to millivolts conversion.
Since wich version that has changed? Seeing the data readed, now it seems that the value readed must be divided into 100, but I'm not sure. It does not seem to be a SIGNED value.

mikeller, 2017-05-17:
I am not an expert in blackbox log viewer myself, but I can have a look if I can find some pointers.

McGiverGim, 2017-05-18:
I'm very new into all of this, so I don't know too much about it. I only want to know what is writed from the firmware. I will search how to solve the problem in the log viewer.
In your PR you said:

> (n.b. The meaning of the amperageLatest value already changed from 'raw ADC reading' to 'centiamp  value` with the introduction of ESC feedback for current sensing.)

So it seems that now, the `amperageLatest` must be divided into 100 (since now it was divided into 10).

If I change this in the blackbox viewer for BF 3.1.7, it seems to work. but I'm not sure because I have not a current meter in my quad and I've only some logs taken from the internet.

Another thing, the amperage latest is a I Field:
```
H Field I name:loopIteration,time,axisP[0],axisP[1],axisP[2],axisI[0],axisI[1],axisI[2],axisD[0],axisD[1],axisD[2],rcCommand[0],rcCommand[1],rcCommand[2],rcCommand[3],vbatLatest,amperageLatest,rssi,gyroADC[0],gyroADC[1],gyroADC[2],debug[0],debug[1],debug[2],debug[3],motor[0],motor[1],motor[2],motor[3]
H Field I signed:0,0,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,1,1,1,1,1,1,1,0,0,0,0
H Field I predictor:0,0,0,0,0,0,0,0,0,0,0,0,0,0,4,9,0,0,0,0,0,0,0,0,0,11,5,5,5
H Field I encoding:1,1,0,0,0,0,0,0,0,0,0,0,0,0,1,3,1,1,0,0,0,0,0,0,0,1,0,0,0
```
I don't know if the signed or encoding values must be changed because now it's a signed value. Now it has a signed=0 (unsigned) and encoding = 1 (unsigned VB). The firmware signed value is an internal value, so maybe it is writed to blackbox as an unsigned, like before, and we don't need to change this fields.
Do you know what is writed to the blackbox exactly? A signed or an unsigned value?

McGiverGim, 2017-05-18:
In the `blackbox.c` it is written as unsigned:
```
{"amperageLatest",-1, UNSIGNED, .Ipredict = PREDICT(0),        .Iencode = ENCODING(UNSIGNED_VB), .Ppredict = PREDICT(PREVIOUS),  .Pencode = ENCODING(TAG8_8SVB), FLIGHT_LOG_FIELD_CONDITION_AMPERAGE_ADC},
```
```
    if (testBlackboxCondition(FLIGHT_LOG_FIELD_CONDITION_AMPERAGE_ADC)) {
        // 12bit value directly from ADC
        blackboxWriteUnsignedVB(blackboxCurrent->amperageLatest);
    }
```
So or this is wrong, or I only need to divide it into 100 and the problem will be solved.

McGiverGim, 2017-05-18:
I've created a PR https://github.com/cleanflight/blackbox-log-viewer/pull/61 to discuss about it.

wolkesson, 2018-05-03:
This doesn't seem to be merged into betaflight/blackbox-log-viewer yet? Same issue as #47.

PR #162 (2018-05-04): From cleanflight: Adjust amperage readings to latest version
Fixes #48.
closed 2018-05-04 by mikeller