betaflight/blackbox-log-viewer #52: Incorrect current value in bfl log exported from firmware 3.1.6

xlla, 2017-03-20:
I am use blackbox explorer v2.5.10.
FC, brf3
battery, 4s
quad, 4 motors

after flash firmware 3.1.6 and restore settings, I found current_meter_scale is missing, so I add current scale in GUI direct, then I found it was renamed to ibat_scale.

I take a test flight, export log, found a very high measured current
[image]
with same parameter in version 3.0.0 , it's correct.
[image]

If I open the old log file generated by Betaflight v3.0.0, it just show correct current value, see second picture I posted.
So, I think Blackbox log viewer is fine, Or there is something wrong in BFL log format, In v3.0.0, exported log was TXT format.

Turismo05, 2017-04-13:
This is the same for me on 3.16. showing over 100amps when it should be around 6.

McGiverGim, 2017-05-29:
As far as I know, the value changed from 3.1.0. The correct value is the actual value divided into 10, so if you have 100 amps, the real value will be 10 amps. The master branch of Cleanflight Blackbox Low Viewer has a correction to show this correctly. 

Can you install manually the master branch of Cleanflight Blackbox Log Viewer to see if the issue is solved?
https://github.com/cleanflight/blackbox-log-viewer

Betaflight Blackbox Log Viewer will surely include this correction in a future, but I don't know when.

xlla, 2017-07-11:
@McGiverGim yes, after install manually the master brach of Betaflight Blackbox Log Viewer, this issue was gone.

closed 2017-07-11 by xlla