betaflight/blackbox-log-viewer #54: Logviewer crashes when I smooth, every time

rodgerramjet, 2017-04-13:
v2.5.11
chrome 57.0.2987.133
windows 10

Steps:
* Open log
* Can click around the timeline thing
* press 's' on keyboard
* app becomes mostly unresponsive- can't click timeline.... some stuff still kinda works but no data is shown on graph
* have to restart to fix

rodgerramjet, 2017-04-13:
[BLACKBOX_LOG_QAVX_20170413_153746.zip](https://github.com/betaflight/blackbox-log-viewer/files/918486/BLACKBOX_LOG_QAVX_20170413_153746.zip)

McGiverGim, 2017-04-17:
I have tried to reproduce your problem without luck. In my computer I can activate/deactivate the smoothing without problem and continue working with your log.
Some idea of what more can I try?

oksada123, 2017-05-15:
Check for "Use hardware acceleration when available" in Chromium-browser settings.

On my workstation (Ubuntu), if this checkbox is not marked, after use of shortcuts "s" and "x" app hangs with error.
[image]

As a workaround you can change settings for every single fields in graphs settings for no smoothing (0%) and no expo (100%) and everything work fine.

rodgerramjet, 2017-05-22:
@oksada123 thanks... I'm on a laptop with a dual graphics card so could be that. I've fixed it by forcing Chrome.exe to use the nvidia card which seems to solve the problem.

Must I close the issue @McGiverGim  ?

mikeller, 2017-05-22:
@rodgerramjet: Yes please.

closed 2017-05-23 by rodgerramjet