betaflight/blackbox-log-viewer #64: Problem setting up the graphics with debug_mode=11 (dfilter)

Juan-ma, 2017-05-25:
I have logged a few flights with debug_mode=dfilter and I can't config any graph in blackbox. I press the Graph setup button, then graph config window appears, but nothing happen when I select any kind of graphic. If I flight debug_mode=notch (or other modes), I can config graphics with no problem.

I have noticed that log file has debug_mode=11 when debug_mode is dfilter. If I change this value with a text editor to other value like 0 or 5, then I can configure the graphics in blackbox without problem.

Thanks!

McGiverGim, 2017-05-26:
There are some PR pending to be included that surely will solve this problem. They're included in the master branch of the Cleanflight Blackbox Explorer. Can you install manually the master branch of Cleanflight Blackbox Explorer to see if it works?

Juan-ma, 2017-05-26:
Thanks McGiverGim,

Tryed with cleanflight blackbox v0.1.6.5 from chrome web store and all of my betaflight log files and it says log file contains no data:

"Error: log is truncated, contains no data"

No idea how to manually install the master branch.

mikeller, 2017-05-27:
Either clone it with git, or download the zip and unzip on your machine. Then in Chrome 'More tools / Extensions' 'Add unpacked extension',  and point it at the top level directory of the repo / unpacked zip.

Juan-ma, 2017-05-27:
Thanks Mikeller and McGiverGim. Cleanfight Blackbox opens the log files correctly.

closed 2017-05-27 by Juan-ma