betaflight/blackbox-log-viewer #65: analyze window wont open

Wtcheesecakes, 2017-05-28:


Davhed, 2017-06-28:
I'm having the same issue. Analyzer button is clickable, but the display does not display on click. 

Using Chrome Version 59.0.3071.115 (Official Build) (64-bit)

ghostface, 2018-03-03:
Same here on chromium with linux

Chromium | 64.0.3282.167 (Developer Build) (64-bit)
-- | --
Revision | bf44778c9ce98ecff1128b225a165728044bbdeb-
OS | Linux
JavaScript | V8 6.4.388.45
Flash | 22.0.0.209 /usr/lib/PepperFlash/libpepflashplayer.so
User Agent | Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/64.0.3282.167 Safari/537.36
Command Line | /usr/lib/chromium/chromium --ppapi-flash-version=22.0.0.209 --ppapi-flash-path=/usr/lib/PepperFlash/libpepflashplayer.so --flag-switches-begin --enable-experimental-web-platform-features --ignore-gpu-blacklist --top-chrome-md=material-hybrid --touch-events=enabled --flag-switches-end

ghostface, 2018-03-03:
I feel stupid - you need to click on the data entry (legend on the right e.g. gyro[roll]) you want to see, then it will show up :)

wolkesson, 2018-04-06:
Could be closed with no action I guess? Add a feature request for better UX in this regard. Should the toolbar button stay hidden until a signal is analysed? Should all signals in the graph be analysed or maybe the first?

PR #189 (2018-05-24): Show the latest field selected (if any) or the first of the list.
Fixes #65.
closed 2018-05-24 by mikeller