betaflight/blackbox-log-viewer #67: Not opening log files, no error message.

packocrayons, 2017-07-06:
Open log file button shows name of file in alt-text, but otherwise the screen does not change.
Latest pull from github and latest release on chrome store, linux mint.

louis-test, 2017-07-31:
I am having the same issue. FC is on 3.1.7.
Notice that it seems to only act up when the black box files exceed a certain size with a single flight. 

Has there been any progress on this issue?

louis-test, 2017-07-31:
I just cloned the repo, and it seems that the latest version is working just fine to open up my logs.

Apologies for the early judgement above. The comment about a single large file w/ one flight was also wrong...

marlboro, 2017-08-01:
BFLV 2.5.11 won't open large logs (say > 6MB) from BF 3.1.7.  If I manually chop up the log file into smaller pieces (say < 4 flights) the open fine.

louis-test, 2017-08-01:
I thought that was the case as well, but I have a ~16MB log that opens fine in the released v2.5.11 with my FC on 3.1.7. Attached are two examples (zip). [16mb log examples.zip](https://github.com/betaflight/blackbox-log-viewer/files/1191990/16mb.log.examples.zip)

I think there might be some interaction with how the new BF Configurator saves larger blackbox logs.

Thoughts?

Seeing as it's working fine in the latest git revision, I might check the commit log to see if there's anything that could hint at what's causing the issue.

louis-test, 2017-08-01:
Okay, I found the 'issue'. It would be nice if we could get the blackbox log viewer to pop up a message saying "unknown filetype" next time....

The new configurator saves the log files as *.BBL ; the old configurator saves the log files as *.BFL.

Here is the commit that enables opening the BBL files. https://github.com/betaflight/blackbox-log-viewer/commit/ecd1dcbc596240023393f0a46637ba74a46a040e

If you just change the extension of your new log files they'll open up fine in the viewer as is...

PR #156 (2018-05-03): Only show extensions we will load in file open dialog
Only these extensions will be loaded .bbl,.txt,.cfl,.bfl,.log,.avi,.mov,.mp4,.mpeg,.json
This PR will only show these in the file open dialog, but the user can still choose to show (and open) any file. Only tested on m$ windows. This could finally fix #67
closed 2018-05-03 by mikeller