betaflight/blackbox-log-viewer #76: Can't setup graphs when using DFILTER debug_mode (3.1.7)

sgcdialler, 2017-07-25:
I have only tried a couple different debug_modes to see whether or not the problem is specific to DFILTER or not. PIDLOOP and NOTCH debug_mode still allow graph setup in BB viewer as expected. With debug_mode set to DFILTER, graph setup displays as a box with 'Save Changes' and 'Cancel' buttons, and the normal button to add graphs (which doesn't do anything).

I saw a recent issue where BB viewer failed to open BF3.2 logs and tried the suggestion to use Cleanflight's viewer, but neither the most recent Chrome app nor the new Windows-installed version could open the log.

McGiverGim, 2017-07-25:
If you want to use the Cleanflight version, you must use the latest git version, manually installing it. The chrome or the install version are far old.

sgcdialler, 2017-07-25:
I tried pulling the latest git version as you suggested and still have issues. I'm attaching the log file I've been using (just change file extension).
[BF-DFILTER.BFL.TXT](https://github.com/betaflight/blackbox-log-viewer/files/1173204/BF-DFILTER.BFL.TXT)

McGiverGim, 2017-07-25:
Yes, I've tested your log and the DEBUG MODE for DFILTER does not work correctly.

I can modify the Blackbox log, but I need this information:

- What is the unit logged by this DEBUG MODE (Hz maybe?)

I've searched the firmware for this DTERM DEBUG MODE and I can't find nothing. Only the declaration :(

I've modified the blackbox and now it shows info, but I don't know if it is correct.
[image]

sgcdialler, 2017-07-25:
That would be my guess? I assumed that the DFILTER debug mode was supposed to be for tuning the d-term filter, just like debug NOTCH is for tuning the gyro notch filters.

sgcdialler, 2017-07-25:
I figured out how to modify the log to open it like you did. It looks to me like it's writing the raw D-term data.

[image]

McGiverGim, 2018-05-04:
I don't know what is the actual state of this, but the DTERM debug mode was removed in latest versions of CF/BF, so maybe is better to close this.

closed 2018-05-29 by McGiverGim