betaflight/blackbox-log-viewer #77: Crash on loading certain logs

thetobby, 2017-08-03:
It crash on loading, I think it might be related to the sampling frequency, Other logs works great. Created with BF 3.2 
[4pidtune.zip](https://github.com/betaflight/blackbox-log-viewer/files/1196285/4pidtune.zip)

thetobby, 2017-08-03:
Changed logging rate from 50% of 2667Hz to 100% and log works, see attached example for working. So I belive the viewer has some issues with that speed. Written to onboard  memory SPRacing F3. 
[4pidtest.zip](https://github.com/betaflight/blackbox-log-viewer/files/1196350/4pidtest.zip)

zytra, 2017-08-25:
Johnny already referenced my issue posted on the BF git.
I have tried multiple logging speed tonight and get the same results regardless of speed and file size.
I have no problem opening similar speed logs from 3.1.7 so I am leaning toward a BF issue; though it could be related to my specific board (BrainFPV RE1).

Cheers.

sepal, 2017-09-16:
As mentioned in the referenced issue I had the same problems with the attached log. This one was done on a kakute F4 with 1.5kHz, the flights were about ~4 to 10 seconds long. The decompressed file has size of about 3.4MB, which seems sketchy. 
I did some debugging and found out that blackbox log viewer hangs in the loop of the countIntentionallySkippedFrames() function flightlog_parser.js:1108. It just runs endlessly there, so probably shouldHaveFrame() never returns true. I don't know how betaflight and the blackbox work, so I'm not sure what the real issue is.
Anyway, commenting the call to countIntentionallySkippedFrames() on flightlog_parser.js:1145 helps,  the viewer was able to display my log this way. I don't know what side effects this hack has though.

[BTFL_BLACKBOX_LOG_20170824_213008.BBL.zip](https://github.com/betaflight/blackbox-log-viewer/files/1308707/BTFL_BLACKBOX_LOG_20170824_213008.BBL.zip)

pulquero, 2017-10-18:
I'm also encountering this problem, tried various logging rates, and file sizes around 200kb. None of them would open. replacing a call to countIntentionallySkippedFrames() with 0 seems to get them to load.

dmitry64, 2017-10-29:
I'm having this exact issue. 
[image]
Here is my log.
[BTFL_BLACKBOX_LOG_20171029_214918_BAD_FILE.BBL.zip](https://github.com/betaflight/blackbox-log-viewer/files/1425435/BTFL_BLACKBOX_LOG_20171029_214918_BAD_FILE.BBL.zip)

richard-scott, 2017-12-10:
I get issues loading log files logged at 1.5kHz, I don't know if it crashes, but the App just hangs. 
See sample good/bad log files on [here](https://github.com/betaflight/betaflight/issues/4728).

haplm, 2018-04-29:
The same problem here. Logged with 1.5kHz, and the viewer hangs when loading the log. 

Is a solution on the horizon? I've got tons of logs with this setting, so I'm just curious if I should redo everything, or wait for a bit...

McGiverGim, 2018-04-29:
The problem was a bug in the firmware and was solved. You can use your log if you edit your file and change the p denom value, the 0 by 1.

haplm, 2018-04-29:
Thanks, but I' seeing 48 in my log file:

H P interval:0/1
H P denom:48
H minthrottle:1070

so I'm not sure what to change. Or could you perhaps point me to the issues in BF that discusses this?

McGiverGim, 2018-04-29:
Was the P interval, my fault. The zero makes an infinite loop. Change 0/1 by 1/1.

McGiverGim, 2018-04-30:
I close this issue, as I said, it was solved in firmware and will be available in the next version.

closed 2018-04-30 by McGiverGim