betaflight/blackbox-log-viewer #81: Info button is missing.

haybeseret, 2017-08-21:
Hi
It seem that there is no more "info" button to show the PIDs and rates configured on the examined log.

Another strange thing, when i manually install 2.5.12 the version shows as 2.5.11, and when installing from the store  it installs 2.5.13, yet stating that 2.5.12 is the latest official.

I have tried to remove all extensions, etc..it did not help.

Thanks,

Chrome 60.0.3112.101 (Official Build) (64-bit)
Windows 10.
Appversion: 2.5.13

Vidalcris, 2017-08-23:
I got exact same problem :-/

Arakon, 2017-08-25:
Use the current version (load as unpacked extension in chrome developer mode) and it will work. This is no longer an issue in the next revision.

closed 2017-08-25 by martinbudden

Vidalcris, 2017-08-25:
load as unpacked extension in chrome developer mode ? ^^
What does that mean ?? why do we have to do that to make it work now ?

Vidalcris, 2017-08-25:
I'm using the latest version and unpacked or not this is not working ..

Vidalcris, 2017-08-25:
`history.pushState is not available in packaged apps.
index.html:1 window.localStorage is not available in packaged apps. Use chrome.storage.local instead.
index.html:1 unload is not available in packaged apps.
flightlog_parser.js:684 Ignoring unsupported header "Log start datetime"
flightlog_parser.js:684 Ignoring unsupported header "P denom"
flightlog_parser.js:684 Ignoring unsupported header "vbat_scale"
flightlog_parser.js:684 Ignoring unsupported header "rc_rate"
flightlog_parser.js:684 Ignoring unsupported header "rc_expo"
flightlog_parser.js:684 Ignoring unsupported header "rc_rate_yaw"
flightlog_parser.js:684 Ignoring unsupported header "rc_expo_yaw"
flightlog_parser.js:684 Ignoring unsupported header "thr_mid"
flightlog_parser.js:684 Ignoring unsupported header "thr_expo"
flightlog_parser.js:684 Ignoring unsupported header "tpa_rate"
flightlog_parser.js:684 Ignoring unsupported header "iterm_windup"
flightlog_parser.js:684 Ignoring unsupported header "vbat_pid_gain"
flightlog_parser.js:684 Ignoring unsupported header "setpoint_relaxation_ratio"
flightlog_parser.js:684 Ignoring unsupported header "dterm_setpoint_weight"
flightlog_parser.js:684 Ignoring unsupported header "acc_limit_yaw"
flightlog_parser.js:684 Ignoring unsupported header "acc_limit"
flightlog_parser.js:684 Ignoring unsupported header "pidsum_limit"
flightlog_parser.js:684 Ignoring unsupported header "pidsum_limit_yaw"
flightlog_parser.js:684 Ignoring unsupported header "gyro_lowpass_type"
flightlog_parser.js:684 Ignoring unsupported header "use_unsynced_pwm"
flightlog_parser.js:684 Ignoring unsupported header "motor_pwm_protocol"
flightlog_parser.js:684 Ignoring unsupported header "dshot_idle_value"
flightlog_parser.js:684 Ignoring unsupported header "Log start datetime"
flightlog_parser.js:684 Ignoring unsupported header "P denom"
flightlog_parser.js:684 Ignoring unsupported header "vbat_scale"
flightlog_parser.js:684 Ignoring unsupported header "rc_rate"
flightlog_parser.js:684 Ignoring unsupported header "rc_expo"
flightlog_parser.js:684 Ignoring unsupported header "rc_rate_yaw"
flightlog_parser.js:684 Ignoring unsupported header "rc_expo_yaw"
flightlog_parser.js:684 Ignoring unsupported header "thr_mid"
flightlog_parser.js:684 Ignoring unsupported header "thr_expo"
flightlog_parser.js:684 Ignoring unsupported header "tpa_rate"
flightlog_parser.js:684 Ignoring unsupported header "iterm_windup"
flightlog_parser.js:684 Ignoring unsupported header "vbat_pid_gain"
flightlog_parser.js:684 Ignoring unsupported header "setpoint_relaxation_ratio"
flightlog_parser.js:684 Ignoring unsupported header "dterm_setpoint_weight"
flightlog_parser.js:684 Ignoring unsupported header "acc_limit_yaw"
flightlog_parser.js:684 Ignoring unsupported header "acc_limit"
flightlog_parser.js:684 Ignoring unsupported header "pidsum_limit"
flightlog_parser.js:684 Ignoring unsupported header "pidsum_limit_yaw"
flightlog_parser.js:684 Ignoring unsupported header "gyro_lowpass_type"
flightlog_parser.js:684 Ignoring unsupported header "use_unsynced_pwm"
flightlog_parser.js:684 Ignoring unsupported header "motor_pwm_protocol"
flightlog_parser.js:684 Ignoring unsupported header "dshot_idle_value"
three.min.js:513 THREE.WebGLRenderer 70`

haybeseret, 2017-08-25:
Did you commited anything ? I recall trying that. I will try again

haybeseret, 2017-08-27:
confirmed it  is working with the current (unpacked) version.