betaflight/blackbox-log-viewer #87: New Window Button doesn't work

TheIsotopes, 2017-12-03:
@McGiverGim @mikeller 
#86 doesn't work correctly ... if i tick the button comes this error

[image]

McGiverGim, 2017-12-03:
Can you give me more info about the problem? Chrome version? Operating system?

TheIsotopes, 2017-12-03:
@McGiverGim using Chrome Version 62.0.3202.94 (64-Bit) on Windows 10 Version 1709 (64-Bit)
same issue also with the cf blackbox-log-viewer

McGiverGim, 2017-12-03:
Strange, I've tested both, BF and CF, Chrome Versión 62.0.3202.94 (64 bits) and Windows 10 1709 (all like you), and they work for me.

We need more people testing this, to see if there's a problem with your system o maybe I made something wrong.

@mikeller can you test this, please?

By the error:

https://github.com/betaflight/blackbox-log-viewer/blob/5b1bd4a58068018f4858864478360277865a178c/js/main.js#L870

The `chrome.app` seems that does not exist in your system.

basdelfos, 2017-12-03:
Just tested it on macOS, works as expected.
Chrome 62.0.3202.94, macOS 10.13.1 (macOS High Sierra).

TheIsotopes, 2017-12-03:
tested now on another system ... Chrome 62.0.3202.94 (64-Bit) on Windows 10 1709 (64-Bit) ... same problem also occurs here

McGiverGim, 2017-12-03:
Maybe you load it in a strange way? Out you simply load it as an unpacked extension?

TheIsotopes, 2017-12-03:
your right ... loaded as unpacked extension and the button works now ... before i'm loaded the index.html in an chrome window ... thx @McGiverGim ... closed now

closed 2017-12-03 by TheIsotopes