betaflight/blackbox-log-viewer #89: Feature request - PID_sum to show effect of pid_sum limits

ctzsnooze, 2017-12-12:
I recently was testing yaw spins and noticed that although I had PID_sum limits enabled in the CLI (as per default), the logs appeared to show no effect of those limits - as if they were not working.

Maybe it would be preferable to display PIDsum *including* the effect of the limit, ie being being capped by the limit?

Seeing the limits - and confirming they are working - will allow the user to better tune them.  It also seems to me to make more sense like this - I'd like the log to show what is actually happening with all the various limits etc.

mikeller, 2017-12-12:
The values after limiting are not logged by the firmware in the first place, which means that the blackbox log viewer can not just display them.
I don't think that re-implementing firmware functionality in the blackbox log viewer in order to be able to calculate derived values (and after all, pretty much all values except for raw sensor input are derived) is the right approach.
For one, this will be prone to be broken whenever the firmware is updated (and the blackbox log viewer forgotten).

It's probably better if you open a feature request against the firmware to have the filtered PIDsum added as a debug log option.

ctzsnooze, 2017-12-13:
I wasn't meaning to change the blackbox viewer at all, but to change what was sent to the viewer by betaflight.  Sorry if I posted in the wrong place.  If you could point me in the direction and how to say it, that would be great.  Do you agree the limited (actual working) value should be the default, not the debug option?

mikeller, 2017-12-13:
No problem: https://github.com/betaflight/betaflight/issues

Yeah, I guess having the limited value (that is actually used) reported as default makes sense.

ctzsnooze, 2018-05-12:
bump
should be fixed.  
if PIDsum is actually limited, what appears in blackbox should be limited.  It should be the actual value the FC is working with.3

McGiverGim, 2018-05-13:
The Blackbox only shows what is in the logs. If the value is not correct, is the firmware who is sending the bad value and must be fixed there. At least this is my opinion.

ctzsnooze, 2018-05-29:
Unfortunately I an not a good enough coder to know how to fix it within betaflight code, if you know what should be done, I'd be very grateful if you could generate the PR.

McGiverGim, 2018-05-29:
If I'm not wrong, you are saying that the limits are ignored. The blackbox shows what the firmware writes, so or the limit is really not working, or the firmware is writing erroneous values. Can you attach a log with the problem? But I think we will need to write the issue in the betaflight firmware github.

ctzsnooze, 2018-05-29:
I don't have a log now, I'll make one.  I just recall making a test where I set a limit to a low value, made a log, and found that the log showed the value exceeding the limit.  It could be that the limit didn't work at all, but I think more likely it depended where blackbox got the value from - the value that went into the log.  And that I couldn't figure out.  :-)  But I think everyone is too busy for this right now  :-)

McGiverGim, 2018-05-30:
If it serves at you, in one test with flint, we add a limit to the iterm and it worked. And in the blackbox the iterm graph was limited by this iterm. I don't know if the bug can be in others.

closed 2018-07-06 by stale[bot]