betaflight/blackbox-log-viewer #96: Q: Any plans for a Standalone App?

richard-scott, 2018-01-08:
With the demise of Chrome Extensions, what plans does this project have to migrate over to a Standalone App?

mikeller, 2018-01-08:
Already done: #88.

richard-scott, 2018-01-08:
Great, where can I download it from?

richard-scott, 2018-01-08:
I've tried building from source with the same gulp commands I use for Betaflight Configurator or BLHeli and I do not get the expected executable created. No errors, just no App.

mikeller, 2018-01-08:
https://github.com/betaflight/blackbox-log-viewer . You'll have to build it yourself for now, we have not yet done a release.

I think there might be a problem with how gulp is handling concurrent tasks in our build processes. We're currently looking into this in the betaflight-configurator (betaflight/betaflight-configurator#830). Once this is sorted, we'll apply the learnings here.

richard-scott, 2018-01-08:
ok, thanks for the info.

closed 2018-01-08 by richard-scott

basdelfos, 2018-01-08:
Yes, I was planning to do the same fix here when betaflight/betaflight-configurator#830 is ironed out (same for BLHeli configurator)