betaflight/blackbox-log-viewer #98: BF 3.3.0 DEBUG_MODE still mismatched

fujin, 2018-01-17:
Hi there!

It seems like blackbox logs generated with the current version of bf/master, with debug_mode set to say `FFT`, are coming through out-of-sync with the debug enum: https://github.com/betaflight/betaflight/blob/master/src/main/build/debug.h#L49-L77

Attached log that was recorded with `debug_mode = FFT`, but appears as `DEBUG_ALTITUDE`. I made an ineffective attempt at modifying the flightlog fielddefs: https://github.com/McGiverGim/blackbox-log-viewer/blob/bcaea907331b731b1c7d2d2d33a16d7a4394ee50/js/flightlog_fielddefs.js#L161-L186 to suit; was unable to get them to sync up. Seems like certain debug modes aren't logged, or able to be handled by blackbox-log-viewer, or similar.

maybe related to: https://github.com/betaflight/betaflight/pull/4910 and https://github.com/betaflight/blackbox-log-viewer/pull/97

[bbl-debug-mode-mismatch.zip](https://github.com/betaflight/blackbox-log-viewer/files/1637545/bbl-debug-mode-mismatch.zip)

GateslayerFPV, 2018-01-17:
I'm getting the same thing as well... but slightly different... FFT = Stack for me. NOTCH = debug for me. Here are some logs both FFT and NOTCH for debug...

[Debug Notch and FFT.zip](https://github.com/betaflight/blackbox-log-viewer/files/1637890/Debug.Notch.and.FFT.zip)

McGiverGim, 2018-01-17:
I will look into it. Thanks! Can you confirm me on what commit of the firmware are you?

GateslayerFPV, 2018-01-17:
I'm not sure... it's from the stable branch and it's whatever @fujin cooked up for me.

fujin, 2018-01-17:
We're livin' off a branch for now: https://github.com/fujin/betaflight/commits/rs2k-stable-alpha-216mHz-OC

McGiverGim, 2018-01-17:
Fixed. You can download the code from here: https://github.com/McGiverGim/blackbox-log-viewer/tree/bf-fix_unused_debug_modes

Or wait until it is merged.

closed 2018-01-17 by mikeller