betaflight/blackbox-log-viewer #101: avcodec_receive_frame could not be located in nw.dll

richard-scott, 2018-01-25:
I've not used the log viewer in a few days, but today's build direct from master gives me this error.

[image]

I'm using the same build process as before which is still working for the configurator.

```
cd blackbox-log-viewer
call git reset --hard
call git pull
call npm install
call gulp apps --win32
```

McGiverGim, 2018-01-25:
What windows version do you have?

richard-scott, 2018-01-25:
Windows 10, 64bit.
Last time I built this looks to have been on the 18th and it all worked then.
I can't build a 64bit version as there is no option for that.

McGiverGim, 2018-01-25:
Yes, you can build 64 bits version. It was merged some days ago. Please test it.

richard-scott, 2018-01-25:
ah it seems that a "git reset --hard" doesn't clean out files created in the last build... I need to use "git clean --force" before my "git pull". All good now :-)

McGiverGim, 2018-01-25:
Is better that you let the issue open. The official version is the 32 bits one. The 64 bits is not officially supported.

richard-scott, 2018-01-25:
The issue is gone now tho... If it comes back I'll open a new case. This one was user error 😀

richard-scott, 2018-04-09:
Ah, this one issue is back but this time I tried with: gulp apps --win32
I'm attempting to setup a new PC to build on.

McGiverGim, 2018-04-09:
Yes, working in a patch. Same issue than here: https://github.com/betaflight/blackbox-log-viewer/issues/130

McGiverGim, 2018-04-09:
Fixed here: https://github.com/betaflight/blackbox-log-viewer/pull/131