betaflight/blackbox-log-viewer #102: Can't open log on standalone explorer

touchthebitum, 2018-02-05:
I can't open log on standalone explorer but on Chrome app it is ok.
My setup blackbox is : 500mhz 
Thanks
[BTFL_BLACKBOX_LOG_Piko_F4_20180204_194406.zip](https://github.com/betaflight/blackbox-log-viewer/files/1696091/BTFL_BLACKBOX_LOG_Piko_F4_20180204_194406.zip)

McGiverGim, 2018-02-05:
I've tried with the standalone version and it has opened without problem. Can you give more info? Operating system? What does it mean "can't open"?

touchthebitum, 2018-02-05:
[image: Immagine incorporata 1]

mikeller, 2018-02-10:
Looks like it's trying to load the log file as video. Update to the latest version and then try again.

touchthebitum, 2018-02-11:
Hello
I can't open this log. I'm using the 3.0.0-RC1. It hangs and I have to close it with ctrl-alt-del.
[BTFL_BLACKBOX_LOG_Kakute_F4_20180211_112324.zip](https://github.com/betaflight/blackbox-log-viewer/files/1714249/BTFL_BLACKBOX_LOG_Kakute_F4_20180211_112324.zip)

touchthebitum, 2018-02-15:
Hello. I'm still  looking for a solution but I noticed that 32khz logs (recorded at 500mhz) are impossible to open but "normal" 8/8khz (recorded at 500mhz) are ok.
The problem is known or I'm the only one with this problem ?
I tried BBE 3.0.0-RC1 and 2.5.9(2.5.12) chrome version, app crash and I have to reboot my chrome browser to load the app again. 
I fly with a Kakute F4 V2 (onboard 16mb flash memory)
Your help is much apreciated.

trendab, 2018-04-04:
I have the same problem. KakuteF4 V2, 8/8 logging at 1khz. File is 10mb, but also smaller ones are not opening. Using 3.0.0 RC1, tried betaflight, cleanflight and the one in the butterflight explorer.
[BTFL_BLACKBOX_LOG_Dragon_#2_20180404_191311.zip](https://github.com/betaflight/blackbox-log-viewer/files/1876904/BTFL_BLACKBOX_LOG_Dragon_.2_20180404_191311.zip)

McGiverGim, 2018-04-09:
@touchthebitum it was a firmware problem. I have uploaded a firmware that solves the problem for KAKUTEF4V2 here: https://github.com/betaflight/betaflight/issues/5625

If you want you can use it until release 3.3.2 of firmware is out.

touchthebitum, 2018-04-09:
Great!! I'll try it. Thanks