betaflight/blackbox-log-viewer #107: debug_mode: for BF3.3 changed from 8 to 6 for NOTCH in blackbox logs

Mechotronic, 2018-02-18:
The debug_mode for the notch type changed from 8 (BF3.2.5) to 6 (BF3.3RC1)in the header for BB logs. I am unable to select the chart type DEBUG_NOTCH in BBE on BF3.3RC1.

There are also some parameter changes that break the header display panel. Some PID values are blank.

Looks like the enumeration is out of order now because the options are in a new order. MIXER and AIRMODE are removed.

#3.3RC1
debug_mode = NONE Allowed values: NONE, CYCLETIME, BATTERY, GYRO, ACCELEROMETER, PIDLOOP, NOTCH, RC_INTERPOLATION, ANGLERATE, ESC_SENSOR, SCHEDULER, STACK, ESC_SENSOR_RPM, ESC_SENSOR_TMP, ALTITUDE, FFT, FFT_TIME, FFT_FREQ, RX_FRSKY_SPI, GYRO_RAW, MAX7456_SIGNAL, MAX7456_SPICLOCK, SBUS, FPORT, RANGEFINDER, RANGEFINDER_QUALITY, LIDAR_TF, CORE_TEMP, RUNAWAY_TAKEOFF

#3.2.5
debug_mode = NONE Allowed values: NONE, CYCLETIME, BATTERY, GYRO, ACCELEROMETER, MIXER, AIRMODE, PIDLOOP, NOTCH, RC_INTERPOLATION, VELOCITY, DFILTER, ANGLERATE, ESC_SENSOR, SCHEDULER, STACK, ESC_SENSOR_RPM, ESC_SENSOR_TMP, ALTITUDE, FFT, FFT_TIME, FFT_FREQ, FRSKY_D_RX, GYRO_RAW

basdelfos, 2018-02-19:
@Mechotronic: did you compile the latest master? The debug_mode is already fixed in #97 to be compatible with BF3.3, but you will need to compile the latest master or wait until the next release of the Blackbox viewer.

You are right about some fields in the header display panel, i noticed that also. The header display panel needs some extra 'love' from us as there are more fields broken.

spatzengr, 2018-02-19:
@basdelfos , i noted the header display panel is much better in CF v3.0.0-RC1.  Maybe can grab that code?

McGiverGim, 2018-02-19:
Yes, some time ago I updated the CF code (it does not have the Header at all). So maybe can be taken from there.

Mechotronic, 2018-02-19:
@basdelfos I have not. I just used the latest available chrome plugin and standalone version. If it has been fixed in a merge I will delete this.

basdelfos, 2018-02-19:
The initial issue you posted (about the debug_mode) is fixed in #97. I have tested that myself. 
Can you create a separate issue about displaying the header fields? After that we can close this issue as it has already been fixed.

closed 2018-02-19 by Mechotronic