betaflight/blackbox-log-viewer #108: Throttle level not represented correctly in some 3D modes

etracer65, 2018-02-18:
Basically it looks like if `feature 3D` is set then the viewer displays the throttle +- the mid_rc point.  The problem is that there are switched modes that change back to normal 1000-2000us throttle ranges.

For BF3.3:

If `3d_switched_mode = OFF` then the 3D disable mode turns off 3D mode (returns to normal throttle range) when the mode is active.

If `3d_switched_mode = ON` the throttle range is always 1000-2000us and the switch 3D mode reverses the motor direction.

McGiverGim, 2018-05-29:
@etracer65 I never used a 3D quad so I don't understand exactly the problem, but if you attach a blackbox log maybe I can look into it.

etracer65, 2018-05-31:
@McGiverGim I'm not a 3D pilot either but noticed the behavior in the logs while working on a 3D related issue.

A little background first (I'm leaving out the 3D deadzones for simplicity).

When ESC's are in 3D mode they apply positive thrust from 50-100% throttle input and negative (reversed) thrust for 50-0% throttle input.

In 3D mode the center throttle position (mid_rc defaulting to 1500) is considered zero throttle.  So at mid_rc the ESC's get the 50% signal and are idling at lowest motor speed.  As the pilot moves the throttle above mid_rc the ESC's see a throttle signal > 50% and spin positively.  When pilot input is below mid_rc the ESC's see less than 50% and reverse direction.

The blackbox representation of this normal 3D mode is fine and there are no problems.

But there are a couple of modifiers to 3D mode that aren't represented correctly.

First is the 3D disable switch mode.  The idea is that when mode is off, the quad behaves like normal 3D - center throttle is zero, above is positive, below is reversed thrust.  When the switch is enabled, then the reversed state is disabled and the pilot's throttle stick command range goes to 1000 to 2000 (using full through), but the motor outputs are scaled from 50-100% using only the positive thrust.  So the quad behaves like a normal non-3D quad (albeit with reduced throttle precision).  This situation is incorrectly represented in the blackbox log.  

[3D disable switch.BFL.zip](https://github.com/betaflight/blackbox-log-viewer/files/2058609/3D.disable.switch.BFL.zip)

In this log the throttle input really ranged from 1000-2000, not 1500-2000.  The motor outputs are correctly represented ad 50-100% (positive 3D mode thrust).

The next case is "3D on a switch".  In this case again the 3D feature is enabled and the ESC's are in 3D mode.  The parameter `3d_switched_mode = ON` makes the "DISABLE / SWITCH 3D" mode turn into the 3D on a switch flavor (rather than disable 3D flavor).  When the mode is off (normal mode), the throttle/motor behavior is identical to previous 3D mode with the disable switch active.  Pilot uses full input 1000-2000 and motors range from 50-100%.  Again the blackbox incorrectly shows the pilot's input ranging from 1500-2000.

[3D on a switch - off.BFL.zip](https://github.com/betaflight/blackbox-log-viewer/files/2058635/3D.on.a.switch.-.off.BFL.zip)

When the 3D on a switch is enabled, then the pilot is controlling the inverted motor thrust.  Pilot input again ranges from 1000-2000, but motors range from 50-0%.  Blackbox shows the pilot input going from 1500-1000 when in reality the stick actually moved from 1000-2000.  Motor outputs are still represented correctly. This is probably the worst one as the throttle stick is represented in reverse of reality.

[3D on a switch - on.BFL.zip](https://github.com/betaflight/blackbox-log-viewer/files/2058645/3D.on.a.switch.-.on.BFL.zip)

I'm not sure this is even completely a BBE problem and possibly the firmware needs to represent the data written to the log more correctly.  And/or BBE might need more information about the 3D settings states to more correctly represent the data.

McGiverGim, 2018-05-31:
Too much info for my brain!!!! :-P 

I will try to test all of them and see if it is a Blackbox or a Firmware issue. The value (1000-2000) is been written by the Firmware, so we can't do too much with it here, and represents the sticks position. The correct is that they always be in the 1000-2000 range, if not I'm pretty sure that the Firmware has wrote it bad (I will check with your files).

But the translation into 0%-100% is being done in the Blackbox. So if this translations from us to % is wrong, and the info about the 3D Mode is at the log too, I can try to fix that.

mikeller, 2018-06-01:
@McGiverGim: Sorry to blow your brain even more.

@etracer65: Yes, in principle, but this is not complete:

> In 3D mode the center throttle position (mid_rc defaulting to 1500) is considered zero throttle. So at mid_rc the ESC's get the 50% signal and are idling at lowest motor speed.

This is wrong. At 1500 in 3D mode the ESC output is off (the same as 1000 in non-3D). The jump from idling at 1500 + 3D-deadband / idling forward to 3D-deadband / idling reverse is entirely done in the firmware, when the stick crosses the 1500 line (there's no way to ever go to 'props are stopped' in there).

Also, take note that when using 3D with Dshot, the negative half is of the ESC input range is reversed: reverse idle is at Dshot range minimum, and reverse full speed is at Dshot range half - 1.

All of this can go a long way to explain the utter mess that is `src/main/flight/mixer.c`.

McGiverGim, 2019-04-10:
Now that the firmware writes in the Blackbox the real sepoint throttle, I think this problem can be fixed in the firmware part. I'm not too sure, but I suppose that now the firmware can write the exact value and we don't need to transform/interpret it.

closed 2026-08-30 by blckmn