betaflight/blackbox-log-viewer #112: Exported video text is super low-resolution and distorted

joshuabardwell, 2018-02-21:
Steps to reproduce: export video. In the resulting .webm file, the text showing the stick positions and frame number is super low-resolution and distorted. I have verified this at both 1080 and 720 export. See screen shot:

[image]

wolkesson, 2018-05-28:
I've recently made some changes to the stick rendering and at least it now works on my log files. @joshuabardwell: Could you please check if the latest version has fixed the problem? You'll have to compile from code. Or if you like, you could post some files here and specify your user and export settings and I'll test it for you.

McGiverGim, 2018-05-28:
You can grab the latest version from here too http://andwho.sytes.net:8080

I think remember this only happens when you have no video, only log. Have you tested it without video @wolkesson ?

wolkesson, 2018-05-28:
The bug is still present when exporting just the log. I’ll look into it!

McGiverGim, 2018-05-28:
This does not seem easy... Good luck! 😊

wolkesson, 2018-05-28:
Hum, it had nothing to do with the canvas size, render quality or resolution (devicePixelRatio nor backingStorePixelRatio). The "export canvas" looked good if added to the visible DOM. Also the resolution of for example the mode text inside the sticks is good. So I suspect it has something to do with webm-writer not liking transperency.

Since my knowledge of webm is limited, I've created a work-around in PR #194 by instead of clearing the canvas (with transperent color) I fill it with black.

PR #194 (2018-05-29): Fix super low resolution export when no video
The webm-writer does not seem like transparent backgrounds. Workaround: Fill the whole canvas with black when exporting without video. Fixes #112.
closed 2018-05-29 by mikeller