betaflight/blackbox-log-viewer #114: Latest commit 156f7e0 won't open blackbox files in web browser any more

ctzsnooze, 2018-03-05:
Is the downloaded zip file no longer supported for running inside web browsers?

This version will open via index.html but when I try to open a blackbox file, nothing happens.  Slightly older version works perfectly for same files.

McGiverGim, 2018-03-05:
No, the structure of the files has been changed, you need to make a `gulp release --chromeos` to generate a version or download it from one of the continuous integration servers (I don't remember the URL now).

McGiverGim, 2018-03-05:
I've found one, test if it works :)
http://andwho.sytes.net:8080/job/Borisb_Blackbox_Log_Viewer/Platform=linux/

McGiverGim, 2018-03-05:
Sorry @ctzsnooze I'm wrong. I didn't remember that the Blackbox contains the old structure (is the configurator the only who has changed).
So this seems a bug.

McGiverGim, 2018-03-05:
How do you load the Blackbox Explorer? Directly opening the HTML file? You must load it as unpacked extension. I did it and it worked. If you load it as unpacked extension, can you attach one of the files that you tried to load?

ctzsnooze, 2018-03-05:
Thanks @McGiverGim  - much appreciated!

In the past I would just download the zip and run index.html in Chrome.  It wasn't essential to load it as an unpacked extension.  

Yes I tried that and it worked.

Maybe the read-me file could be updated to explain this, and include the Jenkins build site, which will be very useful in future.

Sorry for troubling you!

closed 2018-05-19 by stale[bot]