betaflight/blackbox-log-viewer #116: Probably bad frequency meassuring

apoc, 2018-03-10:
There is difference between spectrum analysis and direct measurement by M key, see attached screen shot. In spectrum analysis the motor noise is about 500Hz but if I meassure directly on D-term I get the half (cca 209Hz)

[image]

McGiverGim, 2018-04-30:
There was a bug in the firmware when Gyro loop is different from PID loop. You have more info here: https://github.com/betaflight/blackbox-log-viewer/issues/133

I close this issue. If you can, test the latest master of the firmware to verify this is solved.

closed 2018-04-30 by McGiverGim

apoc, 2018-05-05:
Cool, thank You.