betaflight/blackbox-log-viewer #117: RPMs not shown correct

pango, 2018-03-13:
with debug mode set to "ESC Sensor" Motor RPMs do not get displayed correctly in the current blackbox log viewer. I'd expect rpms go up when I give full throttle. But the log viewer is showing the oposit at the moment. Also the range is weird. It only has values from 1 to 7. 
[image]

pango, 2018-03-13:
here is the whole bblog file: 
[1.5_ESCs_auf_48khz_BTFL_BLACKBOX_LOG_-_20180311_203816.BBL.zip](https://github.com/betaflight/blackbox-log-viewer/files/1807161/1.5_ESCs_auf_48khz_BTFL_BLACKBOX_LOG_-_20180311_203816.BBL.zip)

pango, 2018-03-22:
Let me know If I csn help somehow.

basdelfos, 2018-03-22:
You made the log with the wrong `debug` setting, in the log `debug` is set to `ESC_SENSOR`.
The strange numbers in temperature and rpm are in fact the number of cnc-errors and the data age.

The correct way to log the temperature or the rpm is to set `debug` to `DEBUG_ESC_SENSOR_RPM` or `DEBUG_ESC_SENSOR_TMP` in the CLI. Values `debug[1-4]` are then filled with the temperature or rpm of the correspondig motor number (1-4).

But, actually, there is also a bug in the blackbox viewer:

In the BF firmware the indexes of `debug[]` are filled with the following values, when `debug` is set to `ESC_SENSOR`:
```
enum {
    DEBUG_ESC_MOTOR_INDEX = 0,
    DEBUG_ESC_NUM_TIMEOUTS = 1,
    DEBUG_ESC_NUM_CRC_ERRORS = 2,
    DEBUG_ESC_DATA_AGE = 3,
};
```

Blackbox viewer expects the following when 'debug' is set to `ESC_SENSOR`:
```
'debug[0]':'Motor Index',
'debug[1]':'Timeouts',
'debug[2]':'Temperature',
'debug[3]':'RPM',
```
This should be:
```
'debug[0]':'Motor Index',
'debug[1]':'Timeouts',
'debug[2]':'CNC errors',
'debug[3]':'Data age',
```

commit cc80055a (2018-03-23): Fixes #117
closed 2018-03-23 by mikeller