betaflight/blackbox-log-viewer #119: viewer takes cpu to 100% and fails to open log

shepes, 2018-03-13:
ive tried these log files on 3 different computers with the same result now. running BF 3.3 with a Matek f722 and onboard microsd card. Any attempt to open a log file generated by this FC results in the CPU going to 100% for one core and the log file not opening. Example log file attached, any advise appreciated 
[LOG00007.BFL.zip](https://github.com/betaflight/blackbox-log-viewer/files/1808712/LOG00007.BFL.zip)

mikeller, 2018-03-21:
What version of the log viewer are you using?

shepes, 2018-03-21:
Sorry completely forgot to update this, since formatting the micro SD
card again its been fine since. Initially I was sceptical of this as it
looked like the file was created successfully, passing it through
strings showed at least the header was valid, however clearly something
was up with the formatting. Only difference this time round was I
formatted the card on a windows machine. Previous attempts had been
using OSX. Card was formatted as FAT each time though 

 

Links:
------
[1]
https://github.com/betaflight/blackbox-log-viewer/issues/119#issuecomment-374882421
[2]
https://github.com/

mikeller, 2018-03-21:
@shepes: In general, SD cards used for streaming data (logs, video, photos) should always be formatted with the SD Card Formatter (https://www.sdcard.org/downloads/formatter_4/), and have 'format size adjustment' enabled. This will allow them to be written at the maximum possible speed.

If this is fixed, can you close the issue please?

shepes, 2018-03-21:
Yeah I used that SD Card formatter on my mac and thats what then produced the unreadable logs, after that I formatted it using the windows built in format utility and its been fine since!

mikeller, 2018-03-21:
@shepes: Ok that is weird. But I only ever used the tool on Windows, so I can't comment on there being a problem with the Mac version.