betaflight/blackbox-log-viewer #120: How open or cut large 16MB LOG file?

kszere, 2018-03-17:
I using FC `KAKUTE V2`. Flash size for LOG data is 16MB. When I try opening this file, Chrome Crasching or MSG BOX `Page Unresponsive`.

How cut this file for few smallest?

trendab, 2018-04-03:
Same issue here, same flight controller. Can only open really small log files.

mikeller, 2018-04-04:
More of a firmware problem than a log viewer problem. Can you please close and open in https://github.com/betaflight/betaflight/issues?

McGiverGim, 2018-04-07:
@kszere I've published a test version for this problem in this issue: https://github.com/betaflight/betaflight/issues/5625

McGiverGim, 2018-04-09:
Solved in https://github.com/betaflight/betaflight/pull/5645

I think that this can be closed.

closed 2018-04-09 by McGiverGim