betaflight/blackbox-log-viewer #121: last build shows wrong sample rate

TheIsotopes, 2018-03-21:
@basdelfos since #115 the viewer shows wrong sample rate and choppy lines
analyzer display shows also wrong sample rate ... and since some time the feature `FAILSAFE` in the status bar is displayed incorrectly (see screenshots)... pls can you check this

latest build with #115 
[image]

build before #115 
[image]

basdelfos, 2018-03-22:
Strange, didn't notice that while testing #115, but I'm not sure I really have looked at this.
Can you send the BB log you where using while creating these screenshots?

TheIsotopes, 2018-03-22:
BB log is on the way 

[BTFL_BLACKBOX_LOG_Mark1_20180321_100624.TXT](https://github.com/betaflight/blackbox-log-viewer/files/1836788/BTFL_BLACKBOX_LOG_Mark1_20180321_100624.TXT)

PR #124 (2018-03-23): Fixes sample rate, user settings dialog and ESC_SENSOR debug values
Fixed #121, #123 and #117
closed 2018-03-23 by mikeller