betaflight/blackbox-log-viewer #127: Rebuild the View buttons

wolkesson, 2018-04-06:
The View buttons do not give a good UX, in mine opinion. I would suggest to remove the View dropdown and split it up into button groups; View and Overlay. 

View group would contain:
- Header
- Graph
- Statistics

Overlay ground would contain:
- Video
- Graph
- Model
- Sticks
- Spectrum

Preferences will be moved to a Cog far right. The full-screen button for the spectrum should be moved to the upper right corner of the spectrum display. 

I will provide a Pull Request for this if there's any interest!

wolkesson, 2018-04-10:
To be fair it's actually quite good looking as is! :) So I don't want to do any big changes. But I would like to give room for new signal analyzers, for example spectograms for vibration analysis, transfer functions or step responses for PID tuning. 

Here's sketch: 
[image]
Annotated:
[image]

I'm not sure about the Spectrum Resize button. I'll fade in when you hover the spectrum to allow full screen and normal size. One could argue that it should be a separate view (fullscreen display). I don't see the use case where you look at the spectrum and graph at the same time, since the spectrum is not related to a point in time but rather to the whole file. 

One would need to decide on how to collapse the on smaller displays, but I'll start by hiding the zoom and speed text boxes (not the sliders) and eventually collapse the view and overlay down to a single dropdown. 

Please comment! I'll be my first PR to this project, so I'm not familiar with how you normally operate. Do you have a separate slack (or do you use the betaflight slack?)

wolkesson, 2018-04-11:
Just realized there's a missing `small-screen` definition breaking the responsive layout. There's also a lot of `@media min/max-width`. Is this intentional or should we use bootstraps build in definitions? 
```
// Extra small screen / phone
xs: 0,
// Small screen / phone
sm: 576px,
// Medium screen / tablet
md: 768px,
// Large screen / desktop
lg: 992px,
// Extra large screen / wide desktop
xl: 1200px
```

The downside to using bootstrap's is that you'll have to set classes in the HTML, whereas doing it in the CSS might blow-up the number of definitions.

McGiverGim, 2018-04-13:
I can say... if something is broken, free feel to push a correction. I'm a web developer but my knowledge about responsive layouts is minimum.

wolkesson, 2018-04-17:
PR #140 has been merged meaning this issue can be closed.

closed 2018-04-17 by wolkesson

mikeller, 2018-04-17:
@wolkesson: Protip: If you add `Fixes #<issue number>` as a comment to a pull request, the merge of the pull request will cause the issue to be closed.