betaflight/blackbox-log-viewer #133: Log Viewer v3.0.0 shows a 330hz notch at 660hz

smasslich, 2018-04-13:
D-Term notch filter set at 330hz is showing in log viewer at 660hz.  This is using v3.0.0.  

Screenshot of the issue:
[image]

Log File:
https://www.dropbox.com/s/3vhej8lwsf92gaw/LOG00007.BFL?dl=0

dump file:
https://www.dropbox.com/s/2jt7npkgbkksps3/BTFL_cli_X5_Silky_20180413_000729.txt?dl=0

Thanks!
Steve

McGiverGim, 2018-04-13:
Good catch. I've found the problem/s, but I don't know which solution is best. Is a problem of the firmware, but maybe is better to fix it in the blackbox and the configurator.

flint723, 2018-04-13:
I thought it was a configurator issue at the heart of it (not setting logging rate correctly when gyro/pid rate is unsynced) but I agree the BB Explorer should be fixed to read these type of logs correctly.  Older versions of Explorer work.

McGiverGim, 2018-04-30:
This was solved in the firmware https://github.com/betaflight/betaflight/pull/5688 so I close this.

closed 2018-04-30 by McGiverGim