betaflight/blackbox-log-viewer #137: Default smoothing should be OFF on all channels

ctzsnooze, 2018-04-13:
Most people use blackbox to help with a tune or to help investigate hot motors / assess filtering effectiveness.

Currently the default behaviour is to smooth all traces.

Hence, on power up, any new user will have a display that does not represent the true noise situation of their quadcopter. 

I think it would be very, very much better for the default on all graphs to be with smoothing off.

Then when pressing the 's' key, the display at the bottom should toggle 'smoothed' vs nothing.

Unsmoothed traces are perfectly adequate for PID tuning anyway, so by defaulting to un-smoothed, everyone, or at least 99% of people, will be happy out of the box.

McGiverGim, 2018-05-31:
I think that now this is more or less fixed. 
- It is not off by default, I'm not too sure if this must be. @mikeller what do you think? I prefer to see the traces with smooth but I understand the sense of smooth off. I think that the novice user can prefer smooth activated, it is simpler to see the things. But an advanced user, that looks for spike, noise, and other advanced features can prefer smooth off.
- But now, what you select is remembered for the next time you open the app. So if activated or not by default is less important than before.

Thoughts?

mikeller, 2018-05-31:
I think the remembering of the settings is the important part - a user will set it to whatever they want it to be, and it stays that way.

McGiverGim, 2018-05-31:
True. The default will only happen the first time the user installs the app. So I close this issue. Thanks!

closed 2018-05-31 by McGiverGim