betaflight/blackbox-log-viewer #138: RC Command Graph incorrectly graphs stick position, not RC_Command

ctzsnooze, 2018-04-13:
I noticed that if I put the cursor carefully at the 50% point of an RC Command input (e.g., at the point where the graph of roll is 50% of the way up), the number displayed at the right hand side is *not* as expected.  Instead of seeing 50% of the peak roll rate as commanded, I see much less.  

Eventually I realised that the graph is actually showing RC stick position, not the actual commanded roll rate.  

But most users - myself included - have been comparing the RC command graph to the gyro trace, to assess delay etc.  This has been based on very reasonable assumptions, i.e. that the values at right are not titled 'stick position', nor are they between 0 and 100%, but are expressed in degrees/sec.

I discovered that the setpointRate graphs show the actual RC value that the quad is trying to attain.  

So I think the position would be much, much better for the majority of users if the RC Command graphs showed what setpointRate actually shows.  Then when the graph is at 50%, the number displayed will be 50% of your peak rate, the graph can be directly compared to what the gyro achieves while attempting to track this input signal, and finally, you will see the effect of your expo in the shape of the instructions going to the FC.

So we actually have two waveforms available.  One shows stick position, the other the actual commanded RC rate.  The problem is that their names don't match what they display, and the units are not coherent.

I am of the view that most people did not want to see stick position, and that for RC input they want to see requested roll rate, so they can compare that to gyro.

Hence my recommended solution is:

1.  Leave the name and numerical display of RC Command as it is, but change the graph to show the requested roll rate (i.e. to show the graph currently assigned to setpointRate, and

2.  To provide a graphical representation of crude stick input, re-name setpointRate to RC Stick Position, make the numerics 0-100 for throttle and +/-100 for the others, and make the graph so it shows the current RC command graph.

I'm sure this would be very positively received by most users, and be much clearer overall.

McGiverGim, 2018-05-04:
After reading your text now I understand why the gyro graph is different from the rc graph :)

I'm with you that this must be fixed, but I'm not too sure if your suggested solution is the best. To me, the `RC Command` is more the position of the stick, and maybe to the rate the `setpointRate` will be better understood as `RC Rate`.

So my suggestion will be:
- Let the `RC Command` as the stick position, but change the values in the table to represent that (0-100 for throttle and -100 to +100 in the other axis).
- Rename `setPointRate` to `RC Rate` and let it as is.
- I need to see if the `RC Command` is one of the default graphs. If it is, then is better to change it by the `RC Rate`.

All this without looking at the code, maybe is not as simple as that, but I think it has sense.

@mikeller what do you think about this?

mikeller, 2018-05-04:
@McGiverGim: I am with you, I think separating the two things out clearly is the way to go.

ctzsnooze, 2018-05-04:
Yes agreed, that sounds great!  

So RC Command is sticks (input), and RC Rate is the actual commanded rates.  Perfect.

Thank you!

PS yes RC Command is one of the default graphs, and yes please do change it to RC Rate.  This way users will, in the default screen, compare 'apples to apples' = the set RC rate to the actual gyro rate.

closed 2018-05-12 by mikeller