betaflight/blackbox-log-viewer #142: Any possibility of a linux32 release?

LosD, 2018-04-18:
I know it's ancient, but I have an old netbook which would be perfect for BlackBox Explorer (and BetaFlight Configurator for that matter), except it's running on an old 32-bit Atom processor.

Is there any problems with the linux32 target, since you don't release it, or has it just been a case of "why would anyone ever want this..."? If it's the latter, I'd respectfully request linux32 added to the release targets. If it's the former, I can't really justify the effort (though I could take a look myself, please let me know if you have any pointers).

Anyway, I'll do the release locally tonight, then I guess I'll find out if there are any issues.

LosD, 2018-04-18:
Totally works!

First try was a failure, but that was because I'm using an old Lubuntu (16.04), and I needed to update nsslib.

Thanks again, @AndersHoglund! :)

LosD, 2018-04-18:
(It's incredibly slow, but that's hardly a surprise :))

closed 2018-04-18 by LosD

AndersHoglund, 2018-04-18:
Oooh, yeah. Slow is just the middle name. Sorry, should have warned about that.

LosD, 2018-04-18:
Well, it wasn't exactly unexpected. My test logs are also huuuge (full flights, notch debug, 8khz sample rate).

I expect to use shorter flights for testing filter configurations, and then lower the BB sampling frequency for the PID debugging.

The BetaFlight configurator also works very well, by the way :)

AndersHoglund, 2018-04-18:
Thanks for the reports, and the testing. 
One clarification is needed though. By "Slow" I meant slow download from Jenkins. Did you also mean that, or slow execution of the apps? Or both?

LosD, 2018-04-18:
Oh, I meant the apps (only BB Explorer is slow, but that's probably netbook that's not up to the crunching  of numbers. Configurator is a liiiitle bit slow, but not problematically so)... But yeah, the Jenkins download speed isn't exactly stellar. And for some reason kept failing on my netbook's Firefox, so I downloaded it on my desktop computer and transferred it to the netbook directly.

The download issue might also exist on Chrome, actually; I noticed that Firefox could resume where it failed, so Chrome might just be doing that automatically.