betaflight/blackbox-log-viewer #145: Log will not open.

Strangedave, 2018-04-22:
I tried to open this log file on 2 different pc's and also an older version of black box viewer. I use version 3.3 on another quad and it works fine, however my flash storage on that board is only 2mb. Built a new quad a few days ago, and the board has 16mb of storage. The log files generated at 1.5khz sampling are in the 10 mb range. I have been able to open shorter files, but the larger ones arent working. sample log attached. Its only 5.2mb still not working on my end.

[BTFL_BLACKBOX_LOG_WIN_5_20180422_110416.zip](https://github.com/betaflight/blackbox-log-viewer/files/1935839/BTFL_BLACKBOX_LOG_WIN_5_20180422_110416.zip)

McGiverGim, 2018-04-22:
This was a bug in the firmware, and was solved here: https://github.com/betaflight/betaflight/pull/5645
The patch will be included in Betaflight 3.3.2.

Strangedave, 2018-04-24:
Spoke too soon, Im pretty sure I need to use fpvm betaflight f7. Gonna try
that tomorrow. 

 

Dave G

 

betaflight/betaflight#5645

<https://github.com/betaflight/blackbox-log-viewer/issues/145#issuecomment-3
FRJ04wzks5trLlQgaJpZM4Te7sg> .
<https://github.com/notifications/beacon/Ak1mkIfbmwGiGX3nLuiyp0BQpdFFuGZUks5
trLlQgaJpZM4Te7sg.gif> 

  <http://s.bl-1.com/4hpqQn5.gif>

McGiverGim, 2018-04-30:
@Strangedave Can you confirm this problem was solved and close this issue? Thanks!

Strangedave, 2018-04-30:
Done. Sorry about that. 

Dave Goddard
Garry Real Estate
Office (630)310-8315
Cell (630)992-3283
Email dave@dave-goddard.com
Web www.4ure.com

closed 2018-04-30 by Strangedave