betaflight/blackbox-log-viewer #148: UI setings affects Spectrum analyzer

Legista, 2018-04-24:
When i changing Zoom in Graph setup the peaks in Analyzer changing too. 
I.e., i set 100% zoom - then i have small, normal peaks, when i set 1000% - i have so large peaks that i can't zoom down by vertical slider.
I have this settings on my gyro graph - Smooth 0, Expo 100, Zoom 1000% because i want to see noise RAW. So because of this i need to make another graph with zoom 100 to get normal spectrum data.
If that have been done intentionally you can close the issue. I just thought that no one GUI setting should affect Analyzer data.

ctzsnooze, 2018-05-05:
This is intended behaviour, and useful.  It allows you to vary the vertical scale of the spectrum.

Also note that the amount of flight time included in the spectrum affects how tall the spectrum will be.  Use i and o keys to select the amount of flight time that is the same, when making comparisons.

Legista, 2018-05-17:
Thanks for reply.
I understand - it's  intended behaviour.
So this issue can be closed :)

closed 2018-05-17 by Legista