betaflight/blackbox-log-viewer #150: Cleaning out old and stale branches

wolkesson, 2018-04-25:
Do some house-keeping and delete stale and merged feature branches from back in 2016. This is from before betaflight was forked from cleanflight. These can easily be recreated should a bug be found! This applies to these branches for example: 

- BB-Viewer-Cleanup
- Chart-Grids
- Custom-Graph-Setup
- draggable-graph-fields
- Mousewheel-Controls-Update
- new-layout
- Parser-update
- SpectrumAnalyser
- UI-overhall
- UI-Tidyup
- User-Configurable-Time
- Watermark
- Workspaces

mikeller, 2018-05-04:
Done.

Thanks for bringing this to my attention. I personally think that creating feature branches in project repositories should be avoided unless they are needed for collaboration - the reality is that a lot of the time they end up orphaned, waiting for somebody to put them out of their misery.

wolkesson, 2018-05-04:
Agree! The GitHub-way seems to be to fork and do your branches there. You might as well also collaborate on that fork branch. 

Great clean-up today, both on branches and issues! I'm closing this too now! 8-)

closed 2018-05-04 by wolkesson