betaflight/blackbox-log-viewer #160: BB explorer spectrum analyzer scale reading double

flint723, 2018-05-03:
Just flashed CI build 796, and now (again) the scale in spectrum analyzer reads double what it should.  I know from previous logs the big spike at 625hz is at ~300hz.  The data is correct (and it is a 4khz log), just the scale shown on the spectrum analyzer is wrong.  Not sure what is at fault here.  Log interval appears correctly in the Explorer (1/4 for 16khz pid, 4khz log rate).  I am using unsynced gyro, 32k gyro, 16k pid, Proshot 32k.

I tried on newest master of BB Explorer too, same result.

Here is log:
[flight 91 1000mah.zip](https://github.com/betaflight/blackbox-log-viewer/files/1969407/flight.91.1000mah.zip)
And Diff:
[diff from flight 91.txt](https://github.com/betaflight/blackbox-log-viewer/files/1969408/diff.from.flight.91.txt)

McGiverGim, 2018-05-03:
Strange, I will try to check it... maybe we missed something.

McGiverGim, 2018-05-03:
Fixed in https://github.com/betaflight/blackbox-log-viewer/pull/135 and merged.

closed 2018-05-03 by McGiverGim

flint723, 2018-05-03:
Thanks a lot!